glidernet/python-ogn-client #7: Incorrect parsing of APRS precision and datum option

shortjonescipher, 2016-07-25:
In parse.py, an option like !W53! is matched in parse_aprs() to match.group('latitude_enhancement') = 5 and match.group('longitude_enhancement') = 3. Each of these is effectively the third decimal place of the fractions of a minute in match.group('latitude') and match.group('longitude'), respectively.
But they are each added to the latitude or longitude **after** these have been given the correct sign according to match.group('latitude_sign') or match.group('longitude_sign').
They should have been added **before** the sign adjustment.
The code also has some unnecessary floating-point manipulation which could result in loss of precision. In my local copy I have replaced it as follows:-
```
'latitude': dddmmhhtToDeg('0' + match.group('latitude') + (match.group('latitude_enhancement') or '0')) * (-1 if match.group('latitude_sign') == 'S' else 1),
'longitude':  dddmmhhtToDeg(match.group('longitude') + (match.group('longitude_enhancement') or '0')) * (-1 if match.group('longitude_sign') == 'W' else 1),
```
with a new function in utils.py which replaces dmsToDeg(dms):-
```
def dddmmhhtToDeg(dddmmhht):
    return float(dddmmhht[:3]) + float(dddmmhht[3:]) / 60

```

kerel-fs, 2016-07-31:
Ok. Could you make a pull request? I will push a new release to pip if merged.

shortjonescipher, 2016-08-03:
Sorry, I'm new to this. Just downloaded git for Windows. When I tried to make a pull request, it wanted me to "Compare changes" first. Don't know what I'm supposed to compare with what...

closed 2016-09-01 by kerel-fs

kerel-fs, 2016-09-01:
Thanks! Fixed in d48b3601415474bac678832476f22cb5499f23a9.
Unfortunately I don't know any good resources for 'Git for Windows' but generally I would recommend to read https://git-scm.com/book. Maybe https://help.github.com/articles/creating-a-pull-request/ answers your questions regarding pull requests.

PR #11 (2016-09-01): Add aprs position enhancement before the sign adjustment
Patch provided by shortjonescipher.

Fixes #7.
commit d48b3601 (2016-09-01): Add aprs position enhancement before the sign adjustment

Patch provided by shortjonescipher.

Fixes #7.