glidernet/python-ogn-client #19: Install failure

tobiz, 2017-03-17:
2017-03-17. I tried sudo pip install ogn-client and got:

Downloading/unpacking ogn-client
  Real name of requirement ogn-client is ogn-client
  Downloading ogn-client-0.6.0.tar.gz
  Running setup.py (path:/tmp/pip_build_root/ogn-client/setup.py) egg_info for package ogn-client
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip_build_root/ogn-client/setup.py", line 6, in <module>
        from ogn.client.settings import PACKAGE_VERSION
    ImportError: No module named ogn.client.settings
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip_build_root/ogn-client/setup.py", line 6, in <module>

    from ogn.client.settings import PACKAGE_VERSION

ImportError: No module named ogn.client.settings

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip_build_root/ogn-client
Storing debug log for failure in /home/pjr/.pip/pip.log

Meisterschueler, 2017-03-18:
Please try just "pip install ogn-client". This should work.

Meisterschueler, 2017-03-18:
I cannot reproduce this error. Did you really just typed "sudo pip install ogn-client" ? Could you provide more informations like OS, Python version, ...?

tobiz, 2017-03-18:
Hi,

Yes I really did just type "sudo pip install ogn-client"; that's what I 
normally do if an item is in pypi.  I've not yet investigated getting it 
from github.

The information you requested is:

I have python 2.7.5-5ubuntu installed.

Other details are:

*Operating system: *Ubuntu Linux 14.04.1

*Kernel and CPU: *Linux 3.13.0-113-generic on x86_64

My system is right up to date with the latest fixes etc. as of this email.

I was intending to investigate your s/w for parsing the APRS data rather 
than using my own code.

If there is any other info you'd like let me know.

Rgds

Peter J Robinson

*Email:* pjrobinson@metronet.co.uk
*Sip:*     tobizlinphone1@sip.linphone.org
*Sip:*     tobiz@ekiga.net

tobiz, 2017-03-18:
Hi,

I tried "pip install ogn-client".

I always try this first with any pypi app so as to check it's found 
etc.  On my system this always fails, I assume because of lack of 
privilege hence the addition of sudo.

If you can't see anything wrong I'll try to install from github and let 
you know how I get on.

Rgds

Peter J Robinson

*Email:* pjrobinson@metronet.co.uk
*Sip:*     tobizlinphone1@sip.linphone.org
*Sip:*     tobiz@ekiga.net

tobiz, 2017-03-18:
Hi,

I just thought!

Since I get on running sudo pip install ogn-client:

"from ogn.client.settings import PACKAGE_VERSION

ImportError: No module named ogn.client.settings"

Could it be that on you machine you already have "ogn.client.settings" 
whereas I don't.  Therefore on your machine the pip works but on mine it 
won't.  If this is the case then it suggests ogn.client.settings is 
missing from the package on pypi.

Have you tried the pip install on a machine (or environment) that's 'clean'?

Just a thought.

Peter J Robinson

*Email:* pjrobinson@metronet.co.uk
*Sip:*     tobizlinphone1@sip.linphone.org
*Sip:*     tobiz@ekiga.net

Meisterschueler, 2017-03-18:
Hi Peter,

thanke you for the informations. You are using python 2.7, but ogn-client needs python3.

Best regards
Konstantin

closed 2017-03-18 by Meisterschueler

tobiz, 2017-03-18:
Hi Konstantin

I did think this might be the issue when I looked at the pypi package 
data a while ago.

I might try and build as 2.7 or convert it to 2.7 and see if I can use it.

I'm surprised pip doesn't pick this up as a defined error - would make 
life so much easier!

Thanks for sorting this out.

Rgds

Peter J Robinson

*Email:* pjrobinson@metronet.co.uk
*Sip:*     tobizlinphone1@sip.linphone.org
*Sip:*     tobiz@ekiga.net