glidernet/python-ogn-client #25: Add support for those beacons if required

kerel-fs, 2017-06-04:
During short testing for the next release, I got the following errors due to unsupported APRS/GON messages:
```
Error, This is not a valid OGN message: 9sat/2 104m 1006.1hPa +17.8degC 0% 3.22V 14/-106.5dBm 0/min
Error, This is not a valid OGN message: Webcam: http://www.planeadores.cl
Error, This is not a valid OGN message: El Trocalan hill, Club de Planeadores Vitacura
Error, This is not a valid OGN message: Contact: achanes@example.com, brito.felipe@example.com
```
Do we need support for them?

acasadoalonso, 2017-09-28:
Kerel,
Yes we need to support them, in the OGN version 0.2.6 the received beacon messages are splitter on two messages, one with just the position and the other with the status according with the APRS requirements.
The differences are between the :/ and the :> APRS messages.
AC/.

Meisterschueler, 2017-09-28:
Hello Angel,

line 1 is a tracker string. We should implement a separate tracker parser (work in progress...).
The problem with the APRS comments line 2-4 is: the do not have any pre- or postfix like the other ogn comment parts. So it is impossible to separate corrupted beacons from beacons with webcam or email or owner informations.
If the webcam/email/owner informations looks like `com1"Webcam: http://www.planeadores.cl"` (`com1` for comment no 1., or `ann` for annotation, or ... ... ...) everything would be okay.

acasadoalonso, 2017-09-28:
Hello ...
The comments on the position message are taken from the rtlsdr-ogn.conf
file, for example:

APRS:
{
  Call="ROBLE4";             # APRS callsign (max. 9 characters)
  SAR=2;
  MaxBitError=10;
  Server = [ "190.14.48.221:14580", "glidern3.glidernet.org:14580", "
glidern2.glidernet.org:14580", "glidern1.glidernet.org:14580" ];
  Beacon:
  { Comment = [ "Vitacura Municipal Aerodrome, Club de Planeadores
Vitacurs",
                "Contact: achanes@manquehue.net, brito.felipe@gmail.com ",
                "Antenna: Yagi-directional, on the airfield tower",
                "Webcam: http://www.planeadores.cl"];
  } ;
} ;

So can be any, so perhaps the solution is not to try to parser them, just
pass it as comments.

Cheers.
AC/.

commit 384f0edc (2017-10-02): Fixes #25
closed 2017-10-02 by Meisterschueler