glidernet/python-ogn-client #28: Support for new beacon types

akolar, 2017-09-27:
The Open Glider Network is moving in a direction to support additional devices from different datastreams, as described at [glidernet/ogn-aprs-protocol](https://github.com/glidernet/ogn-aprs-protocol). For this purpose, new beacon types will be introduced and will be encoded as a 8-letter value of the destto field (i.e. the one after the first `>` separator).

We can already see `OGFLR`, `OGNTRK` AND `OGNAVI` messages appear on the APRS stream and other providers will likely follow shortly. Some of those will use a comment format different from what `APRS` messages use.

This will require some changes to be made for this package, as the regex patterns used will fail with the newer formats. Examples of messages that will cause issues are:

```
FLRDDE48A>OGLT24,qAS,LT24:/110453h4030.47N/00338.38W'154/004/A=002253 id25387 +000fpm GPS
NAV07220E>OGNAVI,qAS,NAVITER:/125447h4557.77N/01220.19E'258/056/A=006562 !W76! id1C4007220E +180fpm +0.0rot
FLRDDDD78>OGSKYL,qAS,SKYLINES:/134403h4225.90N/00144.83E'000/000/A=008438 id2816 +000fpm
FLRDDF944>OGSPID,qAS,SPIDER:/212430h3322.02S/07036.14W'281/028/A=004550 id300234010617040 +7dB LWE 3D
ICA3E7540>OGSPOT,qAS,SPOT:/163421h1430.38S/04604.43W'000/000/A=007693 id0-2860357 SPOT3 GOOD
```

As seen from the above examples, parser will be required to choose an appropriate regex pattern based on the value of the destto field in order to parse the comment correctly. Additionally, new function might have to be implemented to properly parse the values encoded in the id fields.
This is why I'd suggest we start discussing possible solutions as soon as possible.

One possible easy fix for the existing implementation would be be to pass the `dstcall` value returned by  the `parse_aprs` function as a second parameter to `parse_ogn_aircraft_beacon`. The former will then use an corresponding regex to parse the comment string.
It would also be possible to wrap the previously mentioned functions together and do everything automagically.

Other solutions are of course possible, but from how I see things, the would require a complete rewrite of the parser.

acasadoalonso, 2017-09-27:
Anze,
First of all thanks to start this discussion.
I wrote a parser in python, I do not use yet this parser yet, I use the
libfap.py module for the time being, however the strategy is the same, I
check the callto parameter and based on that I do the parsing one way or
the other.
All the formats are quite similar, however different, so one parser
function per each callto source is the way to go.
AC/.

Meisterschueler, 2017-09-27:
Hello Anze,

I just started the branch "other_beacon_formats", where the parser_ogn_aircraft_beacon distinguish with the dstcall. Have a look and feel free to implement the naviter parser :)

Viele Grüße
Konstantin

acasadoalonso, 2017-09-27:
Konstantin,
A quick note the format of the message for OGNFLR and OGNTRK is exactly the
same that of current APRS dstcall messages, so you can add it in your "if"
statement.
I will try to submit a PR for the SPOT, SPIDER, LT24, SKYLINES and CAPTURS
format.
And perhaps Anze can submit the version for the NAVITER that is quite
similar to the FLARM ones.
Good nite.
AC/.

acasadoalonso, 2017-09-28:
Update done and PR submitted ... applied at your convenience. AC/.

Meisterschueler, 2017-09-28:
Hello Angel,

thank you for the PR. I have some small comments:

1. since the "old aprs" aircraft beacon regexp has got already a "id...." part, I think we should not ad a second "id...." part just with more digits. We have two ways: a) we make the regex of the first "id...." part weak so it also matches for OGNAVI-1 format or b) we make a new regex for OGNAVI-1. I think this is the more clean solution and with exclusive dstcall we wont have any collision with "old aprs" format
2. do we have a detailled specification for the new formats? If so, then we should write tests for the beacons (like test_naviter_beacon.py)
3. could you please make the pull request against the branch and not the master? While the tests are not green I dont want to merge the code.

Viele Grüße
Konstantin

acasadoalonso, 2017-09-28:
Konstantin,

I defined a new parser type DeviceID on the patterns.py  that is more
generic and no conflicts.
The specs and samples you can find it on:
https://github.com/glidernet/ogn-aprs-protocol
I thought that I made the PR to the new branch, done now with the new
branch. I checked and everything is green.
This branch has no conflicts with the base branch
Cheers.
AC/.

Meisterschueler, 2017-09-28:
Okay, thank you.
@akolar : naviter parser finished. please check and / or enhance

Meisterschueler, 2017-09-30:
- [ ] OGLT24
- [X] OGNAVI
- [ ] OGSKYL
- [ ] OGSPID
- [ ] OGSPOT

akolar, 2017-10-02:
Hi,

first I'm really sorry for the late reply.

>  @akolar : naviter parser finished. please check and / or enhance

I have nothing to add here as you've followed the specification to the letter. Thanks for implementing everything on your own! :)

Anze

Meisterschueler, 2017-10-02:
No Problem: Implementation was simple because of the perfect specification
Viele Grüße
Konstantin

closed 2017-10-02 by Meisterschueler