glidernet/python-ogn-client #46: Again: receiver beacon with description crashes the parser

Meisterschueler, 2018-01-11:
This beacon crashes the parser:

ALFALFAL>APRS,TCPIP*,qAC,GLIDERN4:/221830h3330.40SI07007.88W&/A=008659 Alfalfal Hidroelectric Plant, Club de Planeadores Vitacurs

Traceback (most recent call last):
  File "/Users/konstantin/Development/workspace/python-ogn-client/tests/parser/test_parse.py", line 75, in test_v026_chile
    message2 = parse("VITACURA1>APRS,TCPIP*,qAC,GLIDERN4:/201146h3322.79SI07034.80W&/A=002329 Vitacura Hidroelectric Plant, Club de Planeadores Vitacurs", reference_date=datetime(2015, 1, 1))
  File "/Users/konstantin/Development/workspace/python-ogn-client/ogn/parser/parse.py", line 26, in parse
    message.update(parse_comment(message['comment'], dstcall=message['dstcall'], aprs_type=message['aprs_type']))
  File "/Users/konstantin/Development/workspace/python-ogn-client/ogn/parser/parse.py", line 77, in parse_comment
    return parser.parse(aprs_comment, aprs_type)
  File "/Users/konstantin/Development/workspace/python-ogn-client/ogn/parser/aprs_comment/ogn_parser.py", line 17, in parse
    ac_data = self.parse_aircraft_beacon(aprs_comment)
  File "/Users/konstantin/Development/workspace/python-ogn-client/ogn/parser/aprs_comment/ogn_parser.py", line 34, in parse_aircraft_beacon
    return {'address_type': int(ac_match.group('details'), 16) & 0b00000011,
ValueError: invalid literal for int() with base 16: 'ro'

pjalocha, 2018-01-12:
It catches "id" in "Hidroelectric", you could ask it to catch " id" (a
space before "id"), depending how this is done
or use other smarter ways. Normally after "id" there should be a
hexadecimal number.

Pawel.

Meisterschueler, 2018-01-12:
Thank you, @pjalocha , I changed the regexp for a hex number.

I think the comments in the new beacons (not the dstcall == APRS) should be something like this: "note0:Antenna is 9dB", "note1:Contact: me@ogn.org", "note2:blabla"

commit f562580e (2018-01-12): Merge pull request #47 from Meisterschueler/fix/parser_comment_with_id

Fix/parser comment with id, fixes #46
closed 2018-01-12 by Meisterschueler