glidernet/python-ogn-client #49: Invalid coordinates

Meisterschueler, 2018-01-12:
Sometimes the coordinates leave their valid range: -90<=latitude<=90 or -180<=longitude<=180

`RND000000>APRS,qAS,TROCALAN1:/210042h6505.31S/18136.75W^054/325/A=002591 !W31! idA4000000 +099fpm +1.8rot FL029.04 12.0dB 5e -6.3kHz gps11x17`

1. leave this untouched and let the app handle it ?
2. let the regexp catch only valid coordinates and throw a AprsParseError ?
3. replace the aprs parser part with 'libaprs' library and let it handle the problem ?

pjalocha, 2018-01-12:
This is partly a problem of the OGN receiver... I should fix it not to
allow coordinates out of range.
One can still handle it properly by the interpreter thus either ignore it
or do something else, but not crash anything.

Pawel.

Meisterschueler, 2018-01-13:
Hello Pawel, It crashed my app (ogn-python). Since it is not a valid beacon, I adapted the regex (solution 2.) with caffdabb4e25cb8ce58cb68ecce14ccef91848e9. Now the behaviour is similar to libaprs, which throws a Exception `ParseError("longitutde is out of range (0-180 degrees)"`

I think, long term solution is to replace the APRS part with the libaprs library.

closed 2018-01-13 by Meisterschueler