glidernet/python-ogn-client #63: Add OGNFLYM protocol

Meisterschueler, 2019-06-03:
`FMT898780>OGFLYM,qAS,FLYMASTER:/210139h2915.29N/09609.16W^000/000/A=000072 !W07!` crash the parser.

acasadoalonso, 2019-06-03:
Why ??? it looks OK to me ... my parser seems to handle correctly !!!

Meisterschueler, 2019-06-03:
> Why ??? it looks OK to me ... my parser seems to handle correctly !!!

Because there is no parser for OGFLYM messages.
See here:

```
OgnParseError                             Traceback (most recent call last)
<timed eval> in <module>

<ipython-input-36-b700222ca3c7> in parse_messages2(messages)
      2     result = []
      3     for message in messages:
----> 4         parse(message)
      5     return result

~\AppData\Local\Continuum\anaconda3\lib\site-packages\ogn\parser\parse.py in parse(aprs_message, reference_timestamp)
     26         message.update(parse_comment(message['comment'],
     27                                      dstcall=message['dstcall'],
---> 28                                      aprs_type=message['aprs_type']))
     29     return message
     30 

~\AppData\Local\Continuum\anaconda3\lib\site-packages\ogn\parser\parse.py in parse_comment(aprs_comment, dstcall, aprs_type)
    114         return parser.parse(aprs_comment, aprs_type)
    115     else:
--> 116         raise OgnParseError("No parser for dstcall {} found. APRS comment: {}".format(dstcall, aprs_comment))

OgnParseError: This is not a valid OGN message: No parser for dstcall OGFLYM found. APRS comment: 

```

acasadoalonso, 2019-06-03:
So you need to provide a parser for the OGFLYM messages !!!

Meisterschueler, 2019-06-03:
No, not really. I dont have time for this and I am not interested in these OGFLYM beacons. This is open source, so anyone else who ist interested in these OGFLYM beacons and can code python (parser functions and test cases) can provide a parser.

acasadoalonso, 2019-06-03:
Agree with you !!!

commit b7f51b92 (2019-06-08): Added support for OGFLYM beacons (fixes #63)
closed 2019-06-08 by Meisterschueler