glidernet/python-ogn-client #67: Drop the requirement for some fields in the received position message

pjalocha, 2020-01-09:
Would it be possible to drop the obligation for some fields in the received position message ?
Specifically the number of bit errors corrected, the climb rate and the turn rate.
The reason is that some systems, like PilotAware, don't transmit the latter two numbers and as many messages have no bit errors corrected we can drop then this field to save some space.
Thank you.

pjalocha, 2020-01-09:
As well gpsHxV not all systems are transmitting.

When there is no field saying the number of corrected bits it could be assumed zero.
When climb and/or turn rate are not present, ideally, one would put a flag saying they are absent, as not to mislead the interpretation of the data.

zoranbosnjak, 2020-01-09:
We have observed the same problem.

This is a simple test:
```python
>>> import ogn.parser
>>> ogn.parser.parse('FLR200295>OGFLR,qAS,TT:/071005h4613.92N/01427.53Eg000/000/A=001313 !W00! id1E200295 +000fpm +0.0rot 37.0dB -1.8kHz gps3x5')
Traceback (most recent call last):
...
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
```

This goes without error:
```python
>>> import ogn.parser
>>> ogn.parser.parse('FLR200295>OGFLR,qAS,TT:/071005h4613.92N/01427.53Eg000/000/A=001313 !W00! id1E200295 +000fpm +0.0rot 37.0dB 7e -1.8kHz gps3x5')
{'raw_message': 'FLR200295>OGFLR,qAS,TT:/071005h4613.92N/01427.53Eg000/000/A=001313 !W00! id1E200295 +000fpm +0.0rot 37.0dB 7e -1.8kHz gps3x5', 'reference_timestamp': datetime.datetime(2020, 1, 9, 11, 38, 17, 828743), 'aprs_type': 'position', 'name': 'FLR200295', 'dstcall': 'OGFLR', 'relay': None, 'receiver_name': 'TT', 'timestamp': datetime.datetime(2020, 1, 9, 7, 10, 5), 'latitude': 46.232, 'symboltable': '/', 'longitude': 14.458833333333333, 'symbolcode': 'g', 'track': 0, 'ground_speed': 0.0, 'altitude': 400.2024, 'comment': 'id1E200295 +000fpm +0.0rot 37.0dB 7e -1.8kHz gps3x5', 'address_type': 2, 'aircraft_type': 7, 'stealth': False, 'address': '200295', 'climb_rate': 0.0, 'turn_rate': 0.0, 'signal_quality': 37.0, 'error_count': 7, 'frequency_offset': -1.8, 'gps_quality': {'horizontal': 3, 'vertical': 5}, 'software_version': None, 'hardware_version': None, 'real_address': None, 'signal_power': None, 'beacon_type': 'flarm'}
```

PR #68 (2020-01-15): optional fields in flarm_parser
Make all fields optional. Fixes #67.
commit ab25b2b6 (2020-01-15): optional fields in flarm_parser

Make all fields optional. Fixes #67.
closed 2020-01-15 by Meisterschueler