glidernet/python-ogn-client #70: Unable to run example

vranki, 2020-05-08:
I've got python 3.7 pipenv environment.

I've installed ogn-client with pip. 

Running the second example in README results in:

```python
Traceback (most recent call last):
  File "ogn.py", line 1, in <module>
    from ogn.client import AprsClient
  File "/home/cos/src/ogntest/ogn.py", line 1, in <module>
    from ogn.client import AprsClient
ModuleNotFoundError: No module named 'ogn.client'; 'ogn' is not a package
```
Any ideas?

Meisterschueler, 2020-05-10:
Hi @vranki , could you please double check if you installed ogn-client 0.9.6 correctly?

clementallen, 2020-05-11:
@vranki could this be because your file is also called `ogn.py`?  Perhaps try naming it something different

vranki, 2020-05-13:
Yes, looks like naming the test app ogn.py causes the issue. I got the app to print out messages after renaming the file to something else. Sorry.

closed 2020-05-13 by vranki