glidernet/python-ogn-client #71: Check APRS parser

Meisterschueler, 2020-05-12:
```
ERROR: test_50_live_messages (tests.client.test_AprsClient.AprsClientTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/build/Meisterschueler/python-ogn-client/tests/client/test_AprsClient.py", line 145, in test_50_live_messages
    client.run(callback=process_message, autoreconnect=True)
  File "/home/travis/build/Meisterschueler/python-ogn-client/ogn/client/client.py", line 75, in run
    callback(packet_str)
  File "/home/travis/build/Meisterschueler/python-ogn-client/tests/client/test_AprsClient.py", line 132, in process_message
    message = parse(raw_message)
  File "/home/travis/build/Meisterschueler/python-ogn-client/ogn/parser/parse.py", line 26, in parse
    message = parse_aprs(aprs_message, reference_timestamp)
  File "/home/travis/build/Meisterschueler/python-ogn-client/ogn/parser/parse.py", line 82, in parse_aprs
    raise AprsParseError(message)
nose.proxy.AprsParseError: This is not a valid APRS packet: OGNFD0281>APRS,qAS,NAVITER:/042613h4621.12N/01410.30E'179/-02/A=001640 !W37! id06FD0281 +000fpm +0.0rot
```

acasadoalonso, 2020-05-12:
I found it as well on my logs and I reported that error to the Naviter folks.

AC/.

acasadoalonso, 2020-05-12:
Hi Angel & all,

This data comes from internal testing. We have fixed the problem, this
should not happen again.

Thank you for the heads up, we would never have noticed!

Keep up the great work,
Andrej Kolar
*Enabling Pilot Excellence*
<https://www.naviter.com/>
<https://seeyou.cloud/>  <https://www.soaringspot.com/>
<https://www.facebook.com/naviter.seeyou>
<https://www.instagram.com/naviter.seeyou>
SeeYou runs everywhere now <https://seeyou.cloud/> - learn more
<https://www.naviter.com/2018/11/seeyou-everywhere/>

Meisterschueler, 2020-05-22:
The problem with the negative speed is alive again and flooding my log...

`[2020-05-22 11:07:06,572] ERROR in bulkimport: Parsing error with message: OGNFD0281>APRS,qAS,NAVITER:/090706h4621.06N/01410.29E'179/-02/A=001617 !W64! id06FD0281 +000fpm +0.0rot`

pjalocha, 2020-05-22:
We should maybe speak to Naviter, yesterday we find with one device they
use different ID to report through GSM than through the air.

acasadoalonso, 2020-05-22:
I contacted again to the Naviter folks...
Andrej feel free to use my test APRS for your test, happy to share it.

AC/.

acasadoalonso, 2020-05-22:
Sorry guys. We had a version out that still did that

We have now stopped that at our ogn proxy service.

Please confirm that there are no errors since 1500Z today.

Andrej

Andrej Kolar
--
Enabling pilot excellence
https://www.naviter.com

acasadoalonso, 2020-08-30:
You can close this issue, it is fixed by the naviter folks.

closed 2020-08-30 by Meisterschueler