glidernet/python-ogn-client #78: Error when no bit errors reporting, the 0e indicator

acasadoalonso, 2020-08-08:
The following APRS message reports parser error. 
Parser error: FLRDDE9EF>OGFLR,qAS,LEGY:/161449h4149.71N/00228.15W'266/056/A=005316 !W32! id06DDE9EF +614fpm -0.3rot 10.0dB -7.4kHz gps4x7
Parser error: ICAEFFFED>OGFLR,qAS,LEGY:/160203h4149.48N/00229.45W'281/066/A=006143 !W12! id05EFFFED +396fpm +0.0rot 3.8dB +1.3kHz gps1x2

adding 0e after the dB is OK 

The station is:
LEGY>OGNSDR,TCPIP*,qAC,GLIDERN4:/164546h4149.28NI00228.59W&/A=003360
LEGY>OGNSDR,TCPIP*,qAC,GLIDERN4:>164546h v0.2.7.RPI-GPU CPU:0.3 RAM:516.0/971.1MB NTP:0.5ms/-0.9ppm +81.7C 2/2Acfts[1h] RF:+20+1.5ppm/-0.59dB/+7.1dB@10km[128855]/+9.4dB@10km[3/6]

snip, 2020-08-08:
All additional fields are optional.
Parser should not complain if one is missing.

acasadoalonso, 2020-08-08:
The absence of 0e should means no errors ...

pjalocha, 2020-08-08:
Please make the parser tolerant to the absence of the "0e" and when absent assume error count is zero.

Meisterschueler, 2020-08-29:
Duplicate of #67, solved with #68. Please update your software.

closed 2020-08-29 by Meisterschueler