glidernet/python-ogn-client #81: No need of terminate with forward slash when altitude is not provided

acasadoalonso, 2020-08-30:
It looks that the parser is requiring to terminate the prefix with a forward slash even when altitude is not provided:
For example this APRS message parsers OK:

FNT1118C1>OGNFNT,qAS,BelaVista:/100207h3841.98N/00919.40Wg/ !W72! id3E1118C1 FNT71  25.3dB -13.2kHz

but this one without the / after the coordinates it fails to parser:

FNT1118C1>OGNFNT,qAS,BelaVista:/191934h3841.98N/00919.39Wg !W57! id3E1118C1 FNT71 26.3dB -12.3kHz

Thanks in advance for your great support.
AC/.

Meisterschueler, 2020-08-30:
Both messages should fail:
In the first position message we see a broken APRS part "Altitude in Comment Text" (/A=aaaaaa, where aaaaaa is the altitude in feet). This is clearly a bug in the parser, I will change it.
In the second message the altitude is completely missing... Was that done on purpose?

acasadoalonso, 2020-08-30:
Both message should be ok, it has no altitude information 
First messages parsed Ok, the second message it fails and it should not

Meisterschueler, 2020-08-31:
Can we make any assumptions about messages with no altitude? Is it zero? Or the last valid altitude? According to the sample beacons only capturs and Fanet is injecting position messages without altitude. Maybe we should define that OGN position messages must always be valid 4D so the must have valid lat+lon+alt+time.

snip, 2020-08-31:
No information doesn't mean 0.
If we impose the 4D this means that we will ignore some infos. I think it is better to have incomplet information than no information.

acasadoalonso, 2020-08-31:
In some cases SPOT is not sending the altitude as well.

AC/.

Meisterschueler, 2020-09-01:
The APRS spec says:

> Altitude in Comment Text — The comment may contain an altitude value,
in the form /A=aaaaaa, where aaaaaa is the altitude in feet. For example:
/A=001234. The altitude may appear anywhere in the comment.

So if we allow position messages without altitude, then the second beacon should pass. I will change the parser.
@acasadoalonso : For me the trailing forward slash without altitude seems to be wrong. I did read the spec, but I did not find out how to interpret a trailing forward slash which is in your first position message. Could you point out the part of the spec?

acasadoalonso, 2020-09-01:
I agree that the trailing slash is not need it, as the title of the issue says
I just added that in order to test the parser
I will delete it from the list of valid messages
Thanks for your help

AC/.

commit 0784ecdb (2020-09-01): Fix parser for correct textual altitude (fixes #81)
closed 2020-09-01 by Meisterschueler