glidernet/python-ogn-client #82: Parser should be more specific based on the symbo table

acasadoalonso, 2020-08-30:
The OGN parser could return the aircraft type based on the APRS symbols use to separate latitude and longitude and at the end of the longitude.
It could use the following table:

static const char *AprsIcon[16] = // Icons for various FLARM acftType's
{ "/z",  //  0 = ?
  "/'",  //  1 = (moto-)glider    (most frequent)
  "/'",  //  2 = tow plane        (often)
  "/X",  //  3 = helicopter       (often)
  "/g" , //  4 = parachute        (rare but seen - often mixed with drop plane)
  "\\^", //  5 = drop plane       (seen)
  "/g" , //  6 = hang-glider      (rare but seen)
  "/g" , //  7 = para-glider      (rare but seen)
  "\\^", //  8 = powered aircraft (often)
  "/^",  //  9 = jet aircraft     (rare but seen)
  "/z",  //  A = UFO              (people set for fun)
  "/O",  //  B = balloon          (seen once)
  "/O",  //  C = airship          (seen once)
  "/'",  //  D = UAV              (drones, can become very common)
  "/z",  //  E = ground support   (ground vehicles at airfields)
  "\\n"  //  F = static object    (ground relay ?)
} ;

so for example on this APRS message:

FNT1103CE>OGNFNT,qAS,FNB1103CE:/183736h5057.94N/00801.00Eg354/001/A=001042 !W40! id1E1103CE -02fpm 

the symbols are: / and g so it means a hang glider
on this other APRS message:

FNT1118C1>OGNFNT,qAS,BelaVista:/191929h3841.98N\00919.39Wn !W68! id3E1118C1 FNT71 28.8dB -12.3kHz

the symbols are: \ and n so it means a static object or ground relay station.

Thanks again for the great support.
AC/.

acasadoalonso, 2020-08-31:
I just submitted a PR with the enhancement

Meisterschueler, 2020-08-31:
The parser provides the fields "symboltable" and "symbolcode". These fields are the key for the 192 possible APRS icons. I dont see any advantage to provice for some combinations an english string. Did you also check the performance impact?

acasadoalonso, 2020-09-01:
Almost no impact.
The OGN only have 16 aircraft types, so very fast scan.
I rather provide the english string of those 16 combinations, that every program based on the symbol code and symbol table to build the name of the aircraft type.
Thanks

AC/.

Meisterschueler, 2022-03-11:
Seems to me that there is no exact specification of the meaning of symboltable + symbolcode. It does not make any sense to provide the mapping for just 16 (or 192) combinations if they are not standarized in any way.

closed 2025-04-19 by Meisterschueler