glidernet/python-ogn-client #88: Support for ADSB APRS message

acasadoalonso, 2020-10-02:
We have now support for **ADSB** messages in our OGN APRS network
The APRS message looks like:
ICA4BB26A>OGADSB,qAS,ADSB:/191055h4039.46N\00314.71W^076/309/A=011325 id094BB26A +3136fpm  THY8CV   
ICA406A02>OGADSB,qAS,ADSB:/191055h4023.17N\00427.80W^188/518/A=035025 id09406A02 +064fpm  BAW200M

with a new **TOCALL OGADSB** as described on: https://github.com/glidernet/ogn-aprs-protocol

Meisterschueler, 2022-03-11:
@acasadoalonso : could you please provide the exact specification of the ADSB APRS messages?

acasadoalonso, 2022-03-12:
See the examples:

ICA34614E>OGADSB,qAS,LEMDadsb:/142345h4021.73N\00349.97W^002/071/A=002625 id2534614E -064fpm  0rot fnFYS362    
ICA346316>OGADSB,qAS,LEMDadsb:/142351h4038.45N\00329.05W^043/224/A=010375 id25346316 +3712fpm  -0.19rot fnIBE3066   
ICA3452C8>OGADSB,qAS,LEMDadsb:/142351h4037.79N\00317.99W^126/352/A=012600 id253452C8 +1312fpm  0.0rot fnANE76KX   
ICA495292>OGADSB,qAS,LEMDadsb:/142351h4017.16N\00450.76W^070/410/A=038025 id25495292 -064fpm  0.0rot fnTAP842   regCS-TTR modelA319 
ICA34510F>OGADSB,qAS,LEMDadsb:/142350h4005.41N\00356.12W^074/267/A=009000 id2534510F -1024fpm  0rot fnSWT133P   
ICA4CA4EB>OGADSB,qAS,LEMDadsb:/142351h4034.46N\00315.55W^008/370/A=038000 id254CA4EB +000fpm  0.0rot fnRYR4057  regEI-DPG modelB738 

The std position message, plus fnXXXYYY where XXXYYY is the IATA flight number, plus regZZZZZZ where ZZZZZZ is the registration if known, plus modelXXXX where XXXX is the aircraft model on ICAO code.

Cheers.

AC/.

VirusPilot, 2025-04-14:
Who is in a position to provide the respective `adsb_parser.py`?

acasadoalonso, 2025-04-14:
This will parser the ADSB data from APRS

from ogn.parser import parsefrom datetime import datetime
beacon = parse("ICA342351>OGADSB,qAS,AVX920:/103353h4029.53N/00335.41W^000/000
!W32! id01342351 C0:TXLU13"

,
				reference_timestamp=datetime(2015, 7, 31, 12, 34, 56))

acasadoalonso, 2025-04-22:
It is working OK now

closed 2025-04-22 by acasadoalonso