glidernet/python-ogn-client #106: APRS messages when time is not known

acasadoalonso, 2022-03-08:
Apparently this APRS message is valid:

**OGN87B190>OGNTRK,qAS,OxfBarton:/______h5145.96N/00111.49W'000/001/A=000115 !W08! id0787B190 +000fpm +0.0rot 36.0dB -6.1kHz gps9x14**

on the time are 6 underscores, meaning not time known. 
this is when the OGN-Tracker transmits the last known position, but the position is not current.
we can ignore for tracking purposes, however the parses maks as invalid.

Meisterschueler, 2022-03-09:
This is new for me. @acasadoalonso: could you please indicate where exactly in the specification this is described?

Meisterschueler, 2022-03-12:
Okay, I looked in the APRS specification by myself and found this:

1. reports sent without a timestamp can be regarded as real-time, "current" reports (and the receiving station has to record the time were received), whereas reports sent with a timestamp may or may not be real-time, and may possibly be (very) "old".
2. Stations reporting without a timestamp are generally fixed stations.
3. The APRS Data Type Identifier "/" is reserved for the beacon type "Position **with timestamp**".

So this beacon is by not valid at all.

acasadoalonso, 2022-03-12:
I found exactly the same !!!

AC/.

acasadoalonso, 2022-03-15:
No issue