glidernet/python-ogn-client #107: FANET messages by the OGN tracker

acasadoalonso, 2022-03-08:
This is an example of APRS message generated by the OGN station when it receives a FANET message generated by an OGN tracker

**OGN085B5C>OGNFNT,qAS,GETRONIX:/175419h4804.35N\01444.09En !W88! id3F085B5C FNT71 55.0dB**

it is not OGNTRK on the tocall ID, it is OGNFNT in the tocall ID, other than that the structure is similar to the FNTxxxxxx messages.

Meisterschueler, 2022-03-11:
@acasadoalonso : is there any specification of how these beacons can look like?

acasadoalonso, 2022-03-11:
I showed an example above
The syntax is the same that the FNT messages, the only differences is that are originated by OGN trackers OGNxxxxxx and the “tocall” is OGNFNT

Meisterschueler, 2022-03-11:
No, this is wrong. We see an APRS position message with timestamp and comment. The comment is the part "id3F085B5C FNT71 55.0dB". The regex for the FNT (position) comment is:
`(id(?P<details>[\dA-F]{2})(?P<address>[\dA-F]{6}?)\s?)?(?:(?P<climb_rate>[+-]\d+)fpm)?`
See also here:
https://github.com/glidernet/python-ogn-client/blob/b0f3333fd03a7a48fc46b300c64b003b27950382/ogn/parser/pattern.py#L10
So the line (more exactly the APRS comment) would not pass the FNT messages parser.

acasadoalonso, 2022-03-12:
See the other FANET messages

FNT1118C1>OGNFNT,qAS,BelaVista:>191924h Name="FlrmAIC" 26.0dB -12.1kHz
FNT1118C1>OGNFNT,qAS,BelaVista:/191929h3841.98N\00919.39Wn !W68! id3E1118C1 FNT71 28.8dB -12.3kHz
FNT1118C1>OGNFNT,qAS,BelaVista:/191934h3841.98N\00919.39Wn !W57! id3E1118C1 FNT71 26.3dB -12.3kHz
FNT1118C1>OGNFNT,qAS,BelaVista:/100207h3841.98N/00919.40Wg !W72! id3E1118C1 FNT71  25.3dB -13.2kHz 

it looks quite similar correct ???

AC/.

Meisterschueler, 2022-03-12:
This is correct. They look similar and they have the same undocumented structure. And because of this they also do not pass the parser.

acasadoalonso, 2022-03-12:
The format is:
Examples:
FNT1118C1>OGNFNT,qAS,BelaVista:/191919h3841.98N\00919.39Wn !W68! id3E1118C1
FNT71 26.3dB -12.4kHz FNT1118C1>OGNFNT,qAS,BelaVista:>191924h
Name="FlrmAIC" 26.0dB -12.1kHz
FNT1118C1>OGNFNT,qAS,BelaVista:/191929h3841.98N\00919.39Wn !W68! id3E1118C1
FNT71 28.8dB -12.3kHz The *ID* is *FNTxxxxxx* where the xxxxxx is the FANET
device ID for FANET devices FNT1118C1 or *OGNyyyyyy* where yyyyyy is the
OGN tracker ID for messages generated from OGN trackers devices OGN123456 *The
TOCALL is OGNFNT* The *std position parts*:
FNT1118C1>OGNFNT,qAS,BelaVista:/191919h3841.98N\00919.39Wn
!W68! optionally the *climb rate*: +03fpm plus the *uniqueid*: id3E1118C1
plus the *FANET message FNTxx* where xx is the message ID, FNT71 plus
the *signal
strength and frequency deviation* 28.8dB -12.3kHz Feel free to reach me or
Pawel for further details. AC/.

acasadoalonso, 2022-03-12:
The format is:
Examples:
FNT1118C1>OGNFNT,qAS,BelaVista:/191919h3841.98N\00919.39Wn !W68! id3E1118C1
FNT71 26.3dB -12.4kHz FNT1118C1>OGNFNT,qAS,BelaVista:>191924h
Name="FlrmAIC" 26.0dB -12.1kHz
FNT1118C1>OGNFNT,qAS,BelaVista:/191929h3841.98N\00919.39Wn !W68! id3E1118C1
FNT71 28.8dB -12.3kHz The *ID* is *FNTxxxxxx* where the xxxxxx is the FANET
device ID for FANET devices FNT1118C1 or *OGNyyyyyy* where yyyyyy is the
OGN tracker ID for messages generated from OGN trackers devices OGN123456 *The
TOCALL is OGNFNT* The *std position parts*:
FNT1118C1>OGNFNT,qAS,BelaVista:/191919h3841.98N\00919.39Wn
!W68! optionally the *climb rate*: +03fpm plus the *uniqueid*: id3E1118C1
plus the *FANET message FNTxx* where xx is the message ID, FNT71 plus
the *signal
strength and frequency deviation* 28.8dB -12.3kHz Feel free to reach me or
Pawel for further details.

Meisterschueler, 2025-05-21:
Not relevant any more. New parser does not differentiate between different destination calls

closed 2025-05-21 by Meisterschueler