glidernet/python-ogn-client #108: gps_quality field varies between int, object and string

chrgro, 2022-06-14:
The gps_quality field in the parsed object is occasionally an int (for some message types), and sometimes an object with a horizontal/vertical component, and sometimes a string ("3D"). This makes it hard to map this field into a database for further storing and processing.

A simple search in the repo reveals the issue: https://github.com/glidernet/python-ogn-client/search?q=gps_quality

Could we decide on a single datatype for this field, and modify the others to align with that?

chrgro, 2022-06-14:
I am happy to make a pull request for this once there is some agreement for which datatype to go for :) 

Highlighting the issue: 
`    {
        "_index": "flarm_data",
        "document": {
            "raw_message": "OGN41F247>OGNTRK,qAS,EPPT:>193046h h01 v01 15sat/2/19dB 207m 0.0hPa +21.0degC 52.0% 3.72V 14/-0.0dBm 0/min",
            "reference_timestamp": "2022-06-14 19:30:50.630722",
            "aprs_type": "status",
            "name": "OGN41F247",
            "dstcall": "OGNTRK",
            "receiver_name": "EPPT",
            "timestamp": "2022-06-14 19:30:46",
            "comment": "h01 v01 15sat/2/19dB 207m 0.0hPa +21.0degC 52.0% 3.72V 14/-0.0dBm 0/min",
            "hardware_version": 1,
            "software_version": 1,
            "gps_satellites": 15,
            "gps_quality": 2,
            "beacon_type": "tracker"
        }`

versus
 `{
        "_index": "flarm_data",
        "document": {
            "raw_message": "PAW4067BC>APRS,qAS,PWRossall:/193050h5336.66N\\00235.84W^188/083/A=001350 !W98! id234067BC +000fpm +0.0rot 20.0dB 0e -6.0kHz gps1x1",
            "reference_timestamp": "2022-06-14 19:30:50.630575",
            "aprs_type": "position",
            "name": "PAW4067BC",
            "dstcall": "APRS",
            "relay": null,
            "receiver_name": "PWRossall",
            "timestamp": "2022-06-14 19:30:50",
            "latitude": 53.61115,
            "symboltable": "\\",
            "longitude": -2.5974666666666666,
            "symbolcode": "^",
            "track": 188,
            "ground_speed": 153.70149038807688,
            "altitude": 411.48,
            [......]
            "frequency_offset": -6,
            "gps_quality": {
                "horizontal": 1,
                "vertical": 1
            },
            "beacon_type": "aprs_aircraft",
            "location": [
                -2.5974666666666666,
                53.61115
            ]
        }
    }`

acasadoalonso, 2022-06-14:
you are missing two type of APRS messages, a tracker status report and a position report, two different messages 
check this doc for further details https://github.com/glidernet/ogn-aprs-protocol 

AC/.

Meisterschueler, 2022-06-15:
@chgro: yes, you are right. But this is not an issue you could fix with the parser. The problem is, that there is no single specification for valid gps quality. Some beacons send horizontal and vertial quality, some beacons send integer and some beacons send a string... a quick fix for your software would be str(gps_quality).

closed 2022-06-15 by Meisterschueler

chrgro, 2022-06-17:
I understand that there is no one definition that's shared, but still the parser returns just a plain dict, for all types of messages. It would be useful if this was a consistent dict across the parsers. And if there are different types for e.g. gps_quality across the different messages then that those would be different dict keys in the returned dict. Like gps_quality_direction and gps_quality_scale or something. There is not any consistent metadata in the dict either that enabled introspection as to what the field type would be beyond checking the field itself at runtime. 

I ended up just dropping this field for my application, but I worry about this occurring for other fields too (none have come up so far though).