glidernet/python-ogn-client #114: socket errors with too little traffic

matburnham, 2022-09-10:
Tightly controlled filters seem to pass insufficient traffic to keep things working well.

Running python-ogn-client with a 300 km filter radius around central UK results in minimal traffic, and fairly quickly causes a socket.error (presumably a server disconnect, but it could a client disconnect?)

```
2022-09-10 00:53:01,251 ogn.client.client INFO     Connect to OGN (aprs.glidernet.org/51.68.189.96:14580) as OGNMB with filter: 'r/55.6/-5.5/300'
2022-09-10 00:53:02,218 app               INFO          41/min
2022-09-10 00:53:08,466 ogn.client.client ERROR    socket.error
2022-09-10 00:53:08,676 ogn.client.client INFO     Connect to OGN (aprs.glidernet.org/200.54.220.214:14580) as OGNMB with filter: 'r/55.6/-5.5/300'
2022-09-10 00:53:14,091 ogn.client.client ERROR    socket.error
2022-09-10 00:53:14,126 ogn.client.client INFO     Connect to OGN (aprs.glidernet.org/148.251.228.229:14580) as OGNMB with filter: 'r/55.6/-5.5/300'
```

By increasing the radius of this filter, traffic increases:

```
2022-09-10 00:54:06,769 app               WARNING  Start ogn gateway
2022-09-10 00:54:06,999 ogn.client.client INFO     Connect to OGN (aprs.glidernet.org/200.54.220.214:14580) as OGNMB with filter: 'r/55.6/-5.5/690'
2022-09-10 00:55:00,499 app               INFO         228/min
2022-09-10 00:56:00,352 app               INFO         271/min
2022-09-10 00:57:00,456 app               INFO         276/min
2022-09-10 00:58:00,072 app               INFO         254/min
2022-09-10 00:58:07,241 ogn.client.client INFO     Send keepalive to 200.54.220.214
2022-09-10 00:59:00,063 app               INFO         298/min
```

I don't understand exactly what's happening, although fixing the curt error message in #113 does help. It seems to be timing out, but this is nowhere near the point where a keepalive (with current timing) would make a difference.

matburnham, 2022-09-10:
I wonder if #112 relates at all?

Meisterschueler, 2025-04-19:
#112 is merged (but not released yet). Could you please check if this solves the problem?