glidernet/python-ogn-client #115: No APRS passcode support for APRS write-access

matburnham, 2022-09-10:
python-ogn-client does not support APRS passcode which is required to support APRS write-access.

This is not normally an issue, but does appear to prevent filters based on current location (for example) from working. 

The problem, and expected response from the server could do with documenting, to avoid leading one down a blind alley due to the `unvalidated` response received when passing -1 as the passcode.

Suggested resolutions:

1. allow access to the passcode parameter rather than simply returning -1.

2. Implement the passcode algorithm within the code and (optionally) allow a user to choose -1 or a valid passcode to prevent inadvertent publication of unintended data. This would expose the algorithm, but it is not a well kept secret and offers little security.

Meisterschueler, 2025-04-19:
I was not aware that the client need write-access to get a filtered data stream. Could you please provide a source for this?

Meisterschueler, 2025-04-19:
Just checked a random OGN server for this client. I see unfiltered as well as filtered connections. So for me it seems that write-access is not needed.

14580 | python-ogn-client   1.0 | r/48.536466/9.436244/10   r/43.295728/11.905884/10
-- | -- | --
14580 | python-ogn-client   1.2 | b/ICA4B43AC/ICA4B43AD/ICA4B43AF/ICA4B43B0/ICA4B43B1/ICA4B43B2/ICA4B43C7/ICA4B43CB/ICA4B43CD/ICA4B43CE/ICA4B43CF/ICA4B43D0/ICA4B43D1/ICA4B43D2/ICA4B43D4/ICA4B43D5/ICA4B43D6/ICA4B43D7/ICA4B43E1/ICA4B4333/ICA4B4430/ICA4B4436/ICA4B4484/ICA4B4400/ICA4B435D
14580 | python-ogn-client   1.2 | b/OGN24E224/OGN24E29C/OGN24E298/OGN24E258/ICA31FF15/ICA31FF27/OGN24E244/OGN24E25C/OGN24E260/OGN24E23C/OGN24E278/OGN24E2A0/OGN24E2B0/OGN24E21C/OGN24E248/OGN24E234/OGN24E228/OGN24E3A0/FLRD009D0/FLRDD515C
10152 | python-ogn-client   1.2 |  
10152 | python-ogn-client   1.2 |  
14580 | python-ogn-client   1.2 | r/51.015278/7.006111/100
14580 | python-ogn-client   1.2 |  
10152 | python-ogn-client   1.2 |  
14580 | python-ogn-client   1.2 | r/55.923624/9.755859/195
14580 | python-ogn-client   1.2 | r/50.9/11.7/162

closed 2025-04-19 by Meisterschueler