glidernet/python-ogn-client #119: TelnetParser fails due to new format of output

sfalmo, 2023-06-10:
When using `TelnetParser` on an OGN receiver with v0.2.8 of ogn-rf and ogn-decode, standard messages such as the following cannot be parsed:

```
0.502sec:868.194MHz:   1:2:DDFA2F 163754: [ +49.87356, +12.32815]deg   620m  +4.7m/s  24.4m/s 128.6deg  +3.2deg/s 1m5 02x06m Fn:00f__ -5.60kHz 24.5/35.0dB/0  0e     0.1km 224.3deg +52.2deg
```

This is probably due to a change of ogn-rf or ogn-decode which affected the output format of the telnet stream.
The following regex does not result in a match:
https://github.com/glidernet/python-ogn-client/blob/471840f65798b57429f170ed4e6dcbb69a4ad170/ogn/parser/pattern.py#L131-L154

Some things to note here:
- The `aircraft_type` is actually a hexadecimal digit, not a base 10 int
- The units of `turn_rate` are now written as `deg/s` instead of `deg/sec`
- The fields `channel`, `flarm_timeslot` and `ogn_timeslot` are formatted differently. (I am not exactly sure how to read them, it must be `Fn:00f__` in the above example)
- The last few fields of the regex are optional. I have never observed those fields, so I am not sure about their current format/existence

I know that the telnet output is technically only for debugging, but it would be nice to have a working parser to analyze the local data of an OGN receiver more easily. I will try to make a pull request with an updated regex, but I might need some help for the unknown fields.

snip, 2023-06-10:
As you said this is a debug interface. This should never be used by a software.
Why not connecting to APRS server to get your data?

sfalmo, 2023-06-10:
I have written a small python module ([ogn-alert](https://github.com/sfalmo/ogn-alert)) to geofence and filter OGN data in order to trigger alerts.
In some situations, the TelnetParser could spare an unnecessary round-trip to the APRS server. But you're right, I can also connect to the APRS server to get the data. Is there a filter to receive beacons from only one specific OGN station?

snip, 2023-06-10:
Yes, please check APRS protocol documentation (i don't have it in mind).

closed 2025-03-31 by snip