glidernet/python-ogn-client #126: Naviter parsing failing with 'NoneType' object has no attribute 'group' error

ipearx, 2025-04-07:
For reasons I can't quite figure out, all Naviter data parse function is failing with the error 'NoneType' object has no attribute 'group'. Not sure if Naviter changed something at their end, or what. Been happening a couple of days now at least.

Here's an example of the message being received and the error in my logs:
```
'NoneType' object has no attribute 'group': FLRA51D93>OGNAVI,qAS,NAVITER2:/204507h4444.98N/09323.34W'000/000/A=000925 !W67! id06A51D93 +000fpm +0.0rot
```
Looks like all Naviter traffic stopped being parsed due to this. As a temporary fix I manually replace all 'OGNAVI' to 'GENERIC' in the message string, just to get data flowing again...

snip, 2025-04-07:
It seems Naviter just changed their protocol regarding the id field not following their own protocol description.
https://github.com/glidernet/ogn-aprs-protocol/blob/master/Naviter%20APRS%20format.pdf

@akolar can you please have a look asap so you can restore the normal behavior using the described protocol?
(I fixed the parsing for live.glidernet.org regarding OGNAVI this past weekend)

akolarck, 2025-04-08:
@snip and @ipearx We changed the destto field last week following a request from @acasadoalonso. We now send _destto_ as OGNAVI as per the linked protocol description (which implies version 1).

If you guys have other feedback - it's welcome. Especially with regards to where we don't follow our own protocol description.

snip, 2025-04-08:
Hi @akolarck ,

Here the message received is:
FLRA51D93>OGNAVI,qAS,NAVITER2:/204507h4444.98N/09323.34W'000/000/A=000925 !W67! id06A51D93 +000fpm +0.0rot

So with id field of id06A51D93
Which contains 8 chars, so 32 bits (like OGFLR7) but your protocol version 1 specify 40 bits for this field.

In your protocol description you provided examples like this: id1C4007220E
(So with 40 bits as the description in the document)

Ca nyou pelase check?

Thanks

akolarck, 2025-04-08:
Thanks for this @snip. It's fixed server side and already live.

Two examples:
```
OGNFE0D5A>OGNAVI,qAS,NAVITER2:/094624h4254.51N/00203.62E'163/003/A=001627 !W89! id0006FE0D5A +000fpm +0.0rot
FLRDD82C3>OGNAVI,qAS,NAVITER2:/094139h4529.53N/00909.98E'079/000/A=000427 !W29! id0006DD82C3 +000fpm +0.0rot
```

Please confirm that this works with your parsers now.

acasadoalonso, 2025-04-08:
Naviter has release a new version that conform with their own specs, the ID field is no 10 hex chars instead of 8 hex chars

Python 3.12.3 (main, Feb  4 2025, 14:48:35) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from ogn.parser import parse
>>> from datetime import datetime
>>> beacon = parse("OGNFE0D5A>OGNAVI,qAS,NAVITER2:/094624h4254.51N/00203.62E'163/003/A=001627 !W89! id0006FE0D5A +000fpm +0.0rot",
...                             reference_timestamp=datetime(2015, 7, 31, 12, 34, 56))
>>> print (beacon)
{'raw_message': "OGNFE0D5A>OGNAVI,qAS,NAVITER2:/094624h4254.51N/00203.62E'163/003/A=001627 !W89! **id0006FE0D5A** +000fpm +0.0rot", 'reference_timestamp': datetime.datetime(2015, 7, 31, 12, 34, 56), 'aprs_type': 'position', 'name': 'OGNFE0D5A', 'dstcall': 'OGNAVI', 'relay': None, 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2015, 7, 31, 9, 46, 24), 'latitude': 42.908633333333334, 'symboltable': '/', 'longitude': 2.0604833333333334, 'symbolcode': "'", 'track': 163, 'ground_speed': 5.55547555619555, 'altitude': 495.9096, 'comment': 'id0006FE0D5A +000fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 0, 'address_type': 0, 'reserved': 6, 'address': 'FE0D5A', 'climb_rate': 0.0, 'turn_rate': 0.0, 'beacon_type': 'naviter'}

acasadoalonso, 2025-04-08:
I guess that you can close this issue

ipearx, 2025-04-08:
Yip tested and feeding puretrack.io now. Thanks @snip @akolarck @acasadoalonso

snip, 2025-04-08:
@akolarck can you please check the data you are sending as i'm not sure it is what is intended to be sent about the begining.
Example of what @acasadoalonso parsed: 
```'aircraft_type': 0, 'address_type': 0, 'reserved': 6 ```
Is it what you expected @akolarck ?

Thanks

snip, 2025-04-08:
I'm also still seeing right now the old format message like the following:
```FLRDDEF8B>APRS,qAS,NAVITER:/112923h4403.72N/00559.44E'189/033/A=001503 !W75! id06DDEF8B -060fpm +0.0rot```

Is it expected @akolarck ?

acasadoalonso, 2025-04-08:
There are 2 NAVITER servers, NAVITER and NAVITER2, it looks that it is fixed on NAVITER2, but NOT in NAVITERAC/.Sent from my iPad

snip, 2025-04-08:
> `'aircraft_type': 0, 'address_type': 0, 'reserved': 6 `
> 
> This is incorrect [@acasadoalonso](https://github.com/acasadoalonso) [@snip](https://github.com/snip). 0006 is number according to "device identifier" documentation. Note that the documentation has bits, not bytes. You need to read the digits right-to-left to decode it into bits.
> 

Can you please share what should be the result for the provided example please? So we can fix the parsers.

akolarck, 2025-04-08:
This is becoming fun. Thank you for digging into it. This code was never used until last week, therefore also all of the parsers remain untested.

In our message, bit 1 was always set to 1 by mistake. We fixed and deployed this just now. Bit one is now always zero, as expected and as I clarified above. For gliders you should now be getting  "id0004DD82C3" (note 0004 instead of 0006 as before)

How to convert the id to aircraft type:
[image]

Shift to the right by 2:
[image]

And you get "1" as the aircraft type (glider) for the examples above.

acasadoalonso, 2025-04-08:
Andrej

You are sending on your APRS message 
OGNFE0D5A>OGNAVI,qAS,NAVITER2:/094624h4254.51N/00203.62E'163/003/A=001627 !W89! id0006FE0D5A +000fpm +0.0rot

so the ID is:  **id0006FE0D5A**

perhaps according to your specs it should be: **id0600FE0D5**A

Make sense ???

AC/.

snip, 2025-04-08:
We are currently receiving this format:
```FLR394C76>OGNAVI,qAS,NAVITER2:/145031h4437.52N/00630.37E'029/082/A=007359 !W49! id0004394C76 -060fpm +0.0rot```

With current Python parser we have now:
```
>>> from ogn.parser import parse
>>> beacon = parse(r"FLR394C76>OGNAVI,qAS,NAVITER2:/145031h4437.52N/00630.37E'029/082/A=007359 !W49! id0004394C76 -060fpm +0.0rot")
>>> print(beacon)
{'raw_message': "FLR394C76>OGNAVI,qAS,NAVITER2:/145031h4437.52N/00630.37E'029/082/A=007359 !W49! id0004394C76 -060fpm +0.0rot", 'reference_timestamp': datetime.datetime(2025, 4, 8, 16, 31, 19, 517883), 'aprs_type': 'position', 'name': 'FLR394C76', 'dstcall': 'OGNAVI', 'relay': None, 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2025, 4, 8, 14, 50, 31), 'latitude': 44.6254, 'symboltable': '/', 'longitude': 6.506316666666667, 'symbolcode': "'", 'track': 29, 'ground_speed': 151.84966520267835, 'altitude': 2243.0232, 'comment': 'id0004394C76 -060fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 0, 'address_type': 0, 'reserved': 4, 'address': '394C76', 'climb_rate': -0.3048, 'turn_rate': 0.0, 'beacon_type': 'naviter'}
>>> 
```

So regrading to the fields we are looking about:
```'aircraft_type': 0, 'address_type': 0, 'reserved': 4```

So if i understand it correcly @akolarck it is still not good and we need to review the parser. Is it correct?

If i still understood correcly ```id0004394C76``` sould be parsed as:
```'aircraft_type': 1, 'address_type': 2, 'reserved': 0```

snip, 2025-04-08:
According to the documentatio i understand the following for the 2 firsts bytes of id field:
```
PNTT TTII IIII RRRR
// P => stealth mode
// N => Do not track
// TTTT => aircraftType
// IIIIII => IdType
```

So in my example it should be:
```
PNTT TTII IIII RRRR
0000 0100 0010 0000
```
So  ```0420``` isn't it?

Or we need to swap the 2 first bytes? And you put address_type as 0 (unknown)? But with APRS prefix as FLR?

@akolarck can you please explain a bit?
Thanks

akolarck, 2025-04-08:
@snip 

>So if i understand it correcly @akolarck it is still not good and we need to review the parser. Is it correct?

Yes. 

>PNTT TTII IIII RRRR
0000 0100 0010 0000

It's a number, not a text. The 2^0 digit is on the right side. Correct is

```
RRRR IIII IITT TTNP
0000 0000 0010 0000
```

This will give you

`'aircraft_type': 1, 'address_type': 0, 'reserved': 0`

Hope it's clear now.

snip, 2025-04-08:
> It's a number, not a text. The 2^0 digit is on the right side. Correct is
> 
> ```
> RRRR IIII IITT TTNP
> 0000 0000 0000 0100
> ```

Sorry i can't follow you.
In OGNAVI protocol spec it is written `listed from most to least significant bits`.
So P (stealth) bit should be the most significant bit.
Why putting it on the right side in this case?

srmncnk, 2025-04-10:
@snip @Meisterschueler You are both correct, this is exactly how specs are defined. As @akolarck already said we've been asked to switch from classical APRS-built message which we'd been using up until last week. Now with the switch to our own (OGNAVI) it looks like it was never tested since it had never been used.

I've adapted our messages to honor the specs and deployed it on production (NAVITER2). Please let us know if parsing still fails - I'm happy to do additional modifications if needed. I'll deploy this on the secondary server (NAVITER) as soon as I get the confirmation that everything is okay.

snip, 2025-04-10:
Thanks @srmncnk & @akolarck , it looks much better now.

Some examples:

`
{'raw_message': "FLRDDB1AF>OGNAVI,NAVFE0E69*,qAS,NAVITER2:/164636h4548.52N/00846.37E'000/000/A=000784 !W49! id0420DDB1AF +000fpm +0.0rot", 'reference_timestamp': datetime.datetime(2025, 4, 10, 18, 47, 47, 89010), 'aprs_type': 'position', 'name': 'FLRDDB1AF', 'dstcall': 'OGNAVI', 'relay': 'NAVFE0E69', 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2025, 4, 10, 16, 46, 36), 'latitude': 45.808733333333336, 'symboltable': '/', 'longitude': 8.772983333333332, 'symbolcode': "'", 'track': 0, 'ground_speed': 0.0, 'altitude': 238.9632, 'comment': 'id0420DDB1AF +000fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 1, 'address_type': 2, 'reserved': 0, 'address': 'DDB1AF', 'climb_rate': 0.0, 'turn_rate': 0.0, 'beacon_type': 'naviter'}
`

`
{'raw_message': "NAVFE09A8>OGNAVI,qAS,NAVITER2:/185511h4506.10N/02421.71E'000/000/A=000794 !W47! id0440FE09A8 +000fpm +0.0rot", 'reference_timestamp': datetime.datetime(2025, 4, 10, 18, 55, 41, 621155), 'aprs_type': 'position', 'name': 'NAVFE09A8', 'dstcall': 'OGNAVI', 'relay': None, 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2025, 4, 10, 18, 55, 11), 'latitude': 45.101733333333335, 'symboltable': '/', 'longitude': 24.36195, 'symbolcode': "'", 'track': 0, 'ground_speed': 0.0, 'altitude': 242.0112, 'comment': 'id0440FE09A8 +000fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 1, 'address_type': 4, 'reserved': 0, 'address': 'FE09A8', 'climb_rate': 0.0, 'turn_rate': 0.0, 'beacon_type': 'naviter'}
`

`
{'raw_message': "ICAAA841A>OGNAVI,qAS,NAVITER2:/190810h3221.09N/11106.62W'000/000/A=002146 !W25! id0440AA841A +000fpm +0.0rot", 'reference_timestamp': datetime.datetime(2025, 4, 10, 19, 9, 7, 374872), 'aprs_type': 'position', 'name': 'ICAAA841A', 'dstcall': 'OGNAVI', 'relay': None, 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2025, 4, 10, 19, 8, 10), 'latitude': 32.351533333333336, 'symboltable': '/', 'longitude': -111.11041666666667, 'symbolcode': "'", 'track': 0, 'ground_speed': 0.0, 'altitude': 654.1008, 'comment': 'id0440AA841A +000fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 1, 'address_type': 4, 'reserved': 0, 'address': 'AA841A', 'climb_rate': 0.0, 'turn_rate': 0.0, 'beacon_type': 'naviter'}
`

`
{'raw_message': "FNT0A05FB>OGNAVI,qAS,NAVITER2:/190831h3834.98N/00048.42W'000/000/A=002031 !W65! id1C400A05FB +000fpm +0.0rot", 'reference_timestamp': datetime.datetime(2025, 4, 10, 19, 15, 49, 643433), 'aprs_type': 'position', 'name': 'FNT0A05FB', 'dstcall': 'OGNAVI', 'relay': None, 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2025, 4, 10, 19, 8, 31), 'latitude': 38.5831, 'symboltable': '/', 'longitude': -0.8070833333333333, 'symbolcode': "'", 'track': 0, 'ground_speed': 0.0, 'altitude': 619.0488, 'comment': 'id1C400A05FB +000fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 7, 'address_type': 4, 'reserved': 0, 'address': '0A05FB', 'climb_rate': 0.0, 'turn_rate': 0.0, 'beacon_type': 'naviter'}
`

I'm not sure about the `address_type` when it is not relayed. It looks to be always type 4 (so Naviter adress) even when the prefix is ICA.

srmncnk, 2025-04-11:
@snip Improved handling of relaying. Please confirm it now works for you as expected.

snip, 2025-04-11:
> [@snip](https://github.com/snip) Improved handling of relaying. Please confirm it now works for you as expected.

I'm not sure about what changed regarding relay message:
`
{'raw_message': "FLR115CE1>OGNAVI,NAVFD6611*,qAS,NAVITER2:/133110h4643.49N/00748.78E'213/023/A=007172 !W56! id1C20115CE1 +060fpm +0.0rot", 'reference_timestamp': datetime.datetime(2025, 4, 11, 13, 33, 4, 374008), 'aprs_type': 'position', 'name': 'FLR115CE1', 'dstcall': 'OGNAVI', 'relay': 'NAVFD6611', 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2025, 4, 11, 13, 31, 10), 'latitude': 46.724916666666665, 'symboltable': '/', 'longitude': 7.8131, 'symbolcode': "'", 'track': 213, 'ground_speed': 42.59197926416588, 'altitude': 2186.0256, 'comment': 'id1C20115CE1 +060fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 7, 'address_type': 2, 'reserved': 0, 'address': '115CE1', 'climb_rate': 0.3048, 'turn_rate': 0.0, 'beacon_type': 'naviter'}
`

`address_type` still looks not correct for direct messages:
`
{'raw_message': "FLRDDB1AF>OGNAVI,qAS,NAVITER2:/133111h4600.71N/00930.63E'117/068/A=008950 !W05! id0440DDB1AF +240fpm +0.0rot", 'reference_timestamp': datetime.datetime(2025, 4, 11, 13, 32, 19, 552533), 'aprs_type': 'position', 'name': 'FLRDDB1AF', 'dstcall': 'OGNAVI', 'relay': None, 'receiver_name': 'NAVITER2', 'timestamp': datetime.datetime(2025, 4, 11, 13, 31, 11), 'latitude': 46.011833333333335, 'symboltable': '/', 'longitude': 9.510583333333333, 'symbolcode': "'", 'track': 117, 'ground_speed': 125.92411260709913, 'altitude': 2727.96, 'comment': 'id0440DDB1AF +240fpm +0.0rot', 'stealth': False, 'do_not_track': False, 'aircraft_type': 1, 'address_type': 4, 'reserved': 0, 'address': 'DDB1AF', 'climb_rate': 1.2192, 'turn_rate': 0.0, 'beacon_type': 'naviter'}
`

srmncnk, 2025-04-14:
@snip Regarding relaying in the first raw message:
I added specifically the relayer id (NAVFD6611* - marked as *). So NAVFD6611 is the relayer, FLR115CE1 is the relayee.

Regarding address_type in the second raw message:
Indeed address type was wrong. I deployed support for FLR, ICA & FNT ids - please note that I added a 000101 value for FNT. 

Change in format documented here: https://github.com/naviter/seeyou_file_formats/blob/main/Naviter_APRS_format.md

acasadoalonso, 2025-04-14:
Simon,Do a PR with the new doc pleaseAC/.Sent from my iPad

acasadoalonso, 2025-04-14:
Please use a PR to submit the new documentation to the github
https://github.com/glidernet/ogn-aprs-protocol/pulls

srmncnk, 2025-04-15:
@acasadoalonso @snip https://github.com/glidernet/ogn-aprs-protocol/pull/27. Please confirm you're happy with the live changes.

acasadoalonso, 2025-04-15:
> <https://github.com/snip> glidernet/ogn-aprs-protocol#27

PR merged .... Thx

akolarck, 2025-04-18:
@acasadoalonso I assume you reviewed the changes @srmncnk proposed and that this issue is done for us. We will merge the fixes to our backup server and publish the changes.

Thanks for the collaboration!

commit ebe61f15 (2025-04-19): Catch exception while parsing the APRS comment (fixes #126)
closed 2025-04-19 by Meisterschueler